Gifford47/HCPBridgeMqtt #10: Click Restart Button - Hörmann Error 7

fisch55, 2023-04-08:
As soon as I click the restart button on the website, the connection is interrupted and error message 7 appears on the garage door. The only thing that helps is to switch off the power to the garage door.

Gifford47, 2023-04-14:
which version from which branch do you use?
esp32 or esp8266?
have you already tried the latest dev branch?

if the restart of the esp takes longer than 2 seconds, the bus communication is interrupted and the supramatic throws error 7.

my esp32 setup is running fine for weeks with several restarts.

Gifford47, 2023-04-14:
Please try the lastest development branch. I have merged the last changes from dev to master a few days ago. If you tried the master before that, it could be the problem.

fisch55, 2023-04-14:
Thx for reply. Build a new FW with dev branch an uploaded it to the esp32.

testing the restart button ends in error 7.

which esp32 do you have? Maybe that’s the problem.

Gifford47, 2023-04-15:
I have an esp32 C v4 nodemcu.
Prints it any message on the serial line on reboot? When not you can enable LOG3 debugging in the code and see if there's any additional info

fisch55, 2023-04-15:
I'll take a look into... But before I'll check....If I have no Sensor , do I have to uncomment this Line?

//DS18X20
//#define USE_DS18X20         //Uncomment to use a DS18X20 Sensor (only use one sensor!)
----->>>>>> UNCOMMENT??? const int oneWireBus = 4;     //GPIO where the DS18B20 is connected to

Gifford47, 2023-04-15:
No, only if you want to use one. If you have no sensors you have to disable/comment the global define sensors, if not already happened.

Gifford47, 2023-04-22:
Ok, it seems that's normal during communication abort. The error is saved into supramatic memory and I can't erase it from there. The only option is to reset the volatile memory by restart.
Therefore: only use the restart function for development reasons.

closed 2023-04-22 by Gifford47