Gifford47/HCPBridgeMqtt #11: door open/close acknowledge reliability

Gifford47, 2023-04-28:
sometimes the command for opening or closing the door isn´t executed. there´s no action at the door but on mqtt side comes the last state. i.e.: if the door is closed and it should be opened via mqtt, the command arrives the esp, but nothing happens.

the investigation is;
- secure that the command arrives the esp
- is the esp sending the command to the hcp2 bus?
- is there a delay in the code which denies the execution of the command?
- are the conditions met to execute the command at esp side?

14yannick, 2023-04-28:
What is the status of your Hörmann. Have the motor an error code 4 or 7

Gifford47, 2023-04-29:
It has the normal status. Error 7 is only after a communication loss isn't it?

14yannick, 2023-05-12:
I get regularly an error code 7 and then I have to send to times the command that it works. I'm working adapting the implementation of the modbus communication from this repository.
https://github.com/kfroeschl/HCPBridgeHomeAssistant/tree/rewrite
It works quite well I'm struggling on the setPosition function. I'm experimenting with it since this morning If it work till the end of the weekend I will push an experimental branch in my repository.

14yannick, 2023-05-20:
New code didn't help for me still get error 7 and sometimes I need to send the command two times.
First open command only turn on the light but no movement on the door. on the second command the door start moving.
Did you have observed if the ligh of the Motor turn on when you send your open/close who didn't work?
I have now hooked up a new rs485 ttl from a reputable brand.  Let see if it helps

Gifford47, 2023-05-21:
Yes, my motor has the same reaction. I've never logged the bus, so I think it's the next procedure. I suspect that's I timing issue because of reading and writing the bus at nearly the same time. Or an acknowledgement issue.

14yannick, 2023-05-22:
New RS485 controller didn't help. I'm trying an updated version of https://github.com/emelianov/modbus-esp8266 but I have no great hope that it will fixe my issue.
I have ordered an logic analyser from Aliexpress will try this next.

14yannick, 2023-06-01:
No I didn't try the log3 debugging. It's not only on mqtt commands. When my motor is in code 7 it doesn't matter if the command came from MQTT, webpage or hörmann HSE bisecure handheld. from any source the commande has to be send two times then it works and the error diesapers from the motor. It's like that because of a lost connection the motor goes in protection mode and has to be triggered two times to get reset.

Tysonpower, 2023-06-01:
I used the Code from the master Repo on my ProMatic 4 now and all works fine, it also only blinks when i restart the ESp over the WebUI. After the ESP send a Command (Open/close/ light toggle) the LED stops blinking.
With the For from @14yannick my LED was always blinking and the garage door was acting up a bit, maybe this is also the case with the dev branch, will test that later on my ProMatic 4.

14yannick, 2023-06-01:
After a restart of the esp you should also restart your motor. see here.
https://github.com/Gifford47/HCPBridgeMqtt/issues/10
what did you mean acting up a bit. My fork is came from the Dev Branche from this repository.  In my dev Branche I have took another implementation of the communication with the motor with still the same mqtt functionalities. But didn't solve the issue ¯\_(ツ)_/¯.
https://github.com/14yannick/HCPBridgeMqtt/tree/development

Gifford47, 2023-06-01:
> No I didn't try the log3 debugging. It's not only on mqtt commands. When my motor is in code 7 it doesn't matter if the command came from MQTT, webpage or hörmann HSE bisecure handheld. from any [...]

ok, i will try to rewrite the debug logging that it sends the log to a special mqtt topic. this way we can follow any action on the bus. perhaps we can see the issue at any time ...

Tysonpower, 2023-06-01:
> I used the Code from the master Repo on my ProMatic 4 now and all works fine, it also only blinks when i restart the ESp over the WebUI. After the ESP send a Command (Open/close/ light toggle) the [...]

Update: i now flashed the ndev branch and all works fine. as long as i don't use the restart (what i will never need of course) the ProMatic 4 is not Blinking red and all works as it should. I'm happy for now :)

Gifford47, 2023-06-01:
Please test it for at least 24hours. The issue appears after a longer time some times.

14yannick, 2023-06-07:
I found maybe something. As soon as my ESP32 loose the wifi connection the motor goes to error7.  On the ESP32 there is a esp_reset_reason() 6 ESP_RST_INT_WDTE meaning Reset (software or hardware) due to interrupt watchdog.
I saw that som vtaskdelay are in the wrong place but didn't change the behavior. 
I'm suspecting that maybe the https://github.com/alanswx/ESPAsyncWiFiManager library restart the esp on wifi drop but not sure for now.
Maybe you could also check if you can reproduce the error when you block your ESP in your wifi or turn of Wifi.

fisch55, 2023-06-07:
I can confirm, if WiFi lost , the door goes into error 7 mode.

Tysonpower, 2023-06-07:
Okay good to know, ididn't have any issues the last few Days running it on my Promatic4, but my WIFI in the Garage also is quite stable. So maybe this really is the reason for the issues some users have.

Gifford47, 2023-06-07:
That's good! This is the reason for the communication abort. I think it's easy to fix. The restart really comes from the asyncwifimanager. You can also force to stay offline without a restart/watchdog. But you also drop the AP feature. I will look into it the next days.
The other option would be to remove the manager and use the native esp WiFi support.
Thanks for the investigation!

14yannick, 2023-06-10:
For me the result is the same. As soon I block my ESP32 on the network who cause an connection lost the ESP32 gets restarted. Maybe the error is deeper in the esp32 arduino Framework. Will try with a simple Sketch to see if this happen also.

Gifford47, 2023-06-10:
Ok, I will check if an exception occurs and try to catch the reason...

Gifford47, 2023-06-11:
after some investigation, the following is printed on serial after diconnection from wifi:

```
<\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0> [... cut ...] 0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0><\0>ets Jun  8 2016 00:22:57<\r><\n>
<\r><\n>
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)<\r><\n>
configsip: 0, SPIWP:0xee<\r><\n>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r><\n>
mode:DIO, clock div:1<\r><\n>
load:0x3fff0030,len:1344<\r><\n>
load:0x40078000,len:13864<\r><\n>
load:0x40080400,len:3608<\r><\n>
entry 0x400805f0<\r><\n>
```

going futher ...

14yannick, 2023-06-11:
I have remove all Libs and then add it back. It happen as soon as I include the AsyncMqttClient.

Gifford47, 2023-06-11:
i have the same suspicion. it must have something to do with the async module. it can't handle the disconnection, because the module seems to try to access the MQTT connection, which then no longer exists...

14yannick, 2023-06-11:
This example work without crashing.
https://github.com/OttoWinter/async-mqtt-client/blob/master/examples/FullyFeatured-ESP32/FullyFeatured-ESP32.ino#L23
So I think it should be possible to adjust our coding to make it work.

14yannick, 2023-06-19:
The fix on my dev Branch seems to work. No more restart, no more error 7 and no more acknowledgement issue. Did you gave it a try?

Gifford47, 2023-06-20:
Yes, after 3 days it works without any comm aborts. i`ve forced several wifi reconnects. The issue was solved! Thanks for your investigation, great job! :)

closed 2023-06-20 by Gifford47

14yannick, 2023-06-20:
Should I Prepare a merge for your dev Branche to fix the issue there or are you working on it

Gifford47, 2023-06-20:
i`m preparing a PR for the master in order merge the small changes for openhab.

Gifford47, 2023-07-01:
@14yannick I would like to discuss the future implementations and also if it makes sense to concentrate only one branch/fork because we have mostly the same features but currently diverting code.
Maybe we can discuss it via discord?

14yannick, 2023-07-02:
sure we can check this out on disscord. My username there is nico0.7