Gifford47/HCPBridgeMqtt #18: Bus scan doesn't find anything

dfluegel, 2023-07-15:
**Describe the bug**
Bus scan doesn't finds HCPBridge

**To Reproduce**
Steps to reproduce the behavior:
1. Which motor do you have?
Supramatic E4 (without the DIPs, i've to use the Menu option 37/1 to start a bus scan)

3. Which branch/release did you use?
master branch, 15.07.2022

5. Which peripherals (bme280, HCsr04, ...) did you use?
ESP32 + TTL/UART Converter + 5V Stepdown

7. See error

**Expected behavior**
The Motor doesn't find the HCPBridge.

I've also tried disconnecting the power pins from the RJ12 plug and used the terminals 5 and 20 instead, so the ESP has time too boot before the scan (as suggested by @14yannick in another fork of this project).

The RXD led on the TTL UART converter flashes, but nothing on the TXD led.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Gifford47, 2023-07-15:
Do you see the flashing lights on the rs485 module? Please check the Rx/tx lines between esp32 and rs485.

14yannick, 2023-07-15:
Hi @dfluegel when your esp get power did you see the status update in Homeassistant (even status is unknown). If you don't see it in Homeassistant do you see it on your network and are you able to get to the web interface typing the device ip in your browser. That would already make sure that your esp32 is working.
I assume that you start a bus Scan then you see "b.S." on the screen blinking and then display 0. 
If your esp32 is up and running and Buscan return 0 please send a picture of your build to have a look on it.

dfluegel, 2023-07-16:
Thanks for your answers.

@Gifford47 : I saw the RXD led flashing on my TTL serial to UART converter, the TXD LED wasn't on. As shown on the picute, the RX pin of the TTL RS485/UART converter is connected to RX0 on the ESP32, TX is also connected to TX0 on the ESP32.

I tried to measure with a multimeter, if the pin on the ESP has connection to the pins on the converter - this connection has electrical continuity.

@14yannick : yes, if I'm connecting the ESP (solo) with usb, the state via mqtt changes, also if I'm powering the ESP via the Hörman device and the 24V->5V converter. Yes, I saw b.S and after that 0 - and no voltage on the RJ12 pins.

I'm not sure what you mean with the picture, do you mean this:
[image]

Thanks,

14yannick, 2023-07-16:
ok on the RJ13 cable at your motor which wire color is next to your orange wire(white ore grey) . Could you also add a picture of the backside of your proto boad to be able to follow the trace.

dfluegel, 2023-07-16:
On my RJ12 cables the colors are different, so i'm using the pin numbers from the pictures on the main page.
Yes, I've added the 120 Ohm termination (which has really 117,7 Ohm), here's a better picture of the front and the back:

[image]
[image]

The pins are connected as listed below (I've used a thicker cable because the copper strands of the RJ12 cable are too thin:
6 +
5 deadend
4 white A+, 0 Ohm RJ11 to board
3 yellow B-, 0 Ohm RJ11 to board
2 deadend
1 gnd

[image]

I've also tried switching B- and A+.

dfluegel, 2023-07-16:
As written in my last comment: After switching RX0/TX0 with D16/D17 everything works fine.

closed 2023-07-16 by dfluegel