Gifford47/HCPBridgeMqtt #29: esp32 firmware?

Trilis29, 2023-07-29:
**Describe the bug**
I can only see esp32s3 build flags, how to build for casual espp32?

Trilis29, 2023-07-29:
for line 11 in platformio.ini I have modified
`[env:esp32]`

14yannick, 2023-07-29:
Or use the master branch

Trilis29, 2023-07-30:
Ahhh yes 😀 but stil maybe should be commited so that esp32 goes by default also

Trilis29, 2023-08-02:
> You can just click the default build target, that should use ep32 as well as the sensor only targets without the s3 prefix

If you choose default it builds all except default 😊

jkaberg, 2023-08-03:
> You can just click the default build target, that should use ep32 as well as the sensor only targets without the s3 prefix

Doesn't apply for us that doesn't use VSC 😄 We need to modify the .ini and add an additional environment as it sits, easy fix tho

Tysonpower, 2023-08-03:
Adde the esp32 target back, see Pull Request #35

closed 2023-08-03 by Tysonpower