Gifford47/HCPBridgeMqtt #30: The naming of MQTT entities since Home Assistant 2023.8.0

apromix, 2023-08-03:
Hi everybody!

At first many thanks for this project!

Today I updated HA and saw this warnings:

[image]
[image]

More info here:
https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

So small fix of MQTT entries is needed.

Have a nice day!

shellyeah, 2023-08-03:
Changing the value in [Line 11 in the configuration.h](https://github.com/Gifford47/HCPBridgeMqtt/blob/fe5e4b99b7eb255004075ceb5607656305a45bd6/HCPBridgeESP32/src/configuration.h#L11) (#define DEVICE_ID) to something other than garage_door should do the trick.

14yannick, 2023-08-03:
Temporary fix is in development branch. Has to ne considered in new preferences implementation
https://github.com/Gifford47/HCPBridgeMqtt/discussions/20

closed 2023-08-03 by 14yannick