Gifford47/HCPBridgeMqtt #31: Loss of WIFI results in WIFI connect loop

jkaberg, 2023-08-03:
Hi, whenever I update my (Unifi) AP or otherwise reboot the AP the ESP32 will go into an connection loop (connect and then disconnect shortly after) which is only solved by an hard reboot of the ESP32 (disconnect power, wait 30s and connect it again)

See image (at 11:45 I rebooted the AP):
[image]

Any suggestions? No sensor and standard configuration

Gifford47, 2023-08-03:
Hi! Please use the issue template in order to give us all info we need to investigate this issue.

jkaberg, 2023-08-03:
> Hi! Please use the issue template in order to give us all info we need to investigate this issue.

Done

14yannick, 2023-08-03:
I have an idea what could be the issue. will have a look on it this this evening

14yannick, 2023-08-03:
bugfix have been merged in Dev and waiting for approval to get to master

jkaberg, 2023-08-04:
I just tried to compile latest master and I get this

```
src/main.cpp:120:24: error: no matching function for call to 'WiFiClass::begin(String&, String&)'
   WiFi.begin(ssid, pass);
                        ^
In file included from /root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                 from .pio/libdeps/esp32/AsyncElegantOTA/src/AsyncElegantOTA.h:15,
                 from src/main.cpp:3:
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:47:17: note: candidate: 'wl_status_t WiFiSTAClass::begin(const char*, wpa2_auth_method_t, const char*, const char*, const char*, const char*, const char*, const char*, int32_t, const uint8_t*, bool)'
     wl_status_t begin(const char* wpa2_ssid, wpa2_auth_method_t method, const char* wpa2_identity=NULL, const char* wpa2_username=NULL, const char *wpa2_password=NULL, const char* ca_pem=NULL, const char* client_crt=NULL, const char* client_key=NULL, int32_t channel=0, const uint8_t* bssid=0, bool connect=true);
                 ^~~~~
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:47:17: note:   no known conversion for argument 1 from 'String' to 'const char*'
In file included from /root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                 from .pio/libdeps/esp32/AsyncElegantOTA/src/AsyncElegantOTA.h:15,
                 from src/main.cpp:3:
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:48:17: note: candidate: 'wl_status_t WiFiSTAClass::begin(const char*, const char*, int32_t, const uint8_t*, bool)'
     wl_status_t begin(const char* ssid, const char *passphrase = NULL, int32_t channel = 0, const uint8_t* bssid = NULL, bool connect = true);
                 ^~~~~
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:48:17: note:   no known conversion for argument 1 from 'String' to 'const char*'
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:49:17: note: candidate: 'wl_status_t WiFiSTAClass::begin(char*, char*, int32_t, const uint8_t*, bool)'
     wl_status_t begin(char* ssid, char *passphrase = NULL, int32_t channel = 0, const uint8_t* bssid = NULL, bool connect = true);
                 ^~~~~
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:49:17: note:   no known conversion for argument 1 from 'String' to 'char*'
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:50:17: note: candidate: 'wl_status_t WiFiSTAClass::begin()'
     wl_status_t begin();
                 ^~~~~
/root/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:50:17: note:   candidate expects 0 arguments, 2 provided
Compiling .pio/build/esp32/liba82/DallasTemperature/DallasTemperature.cpp.o
*** [.pio/build/esp32/src/main.cpp.o] Error 1
Compiling .pio/build/esp32/lib767/Wire/Wire.cpp.o
=========================================================================================================================== [FAILED] Took 4.13 seconds ===========================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:04.129
===================================================================================================================== 1 failed, 0 succeeded in 00:00:04.129 =====================================================================================================================
root@dunder:~/dev#
```

I've set `STA_SSID`, `STA_PASSWD` and the mqtt parts as before

14yannick, 2023-08-04:
A lot of code change taking place at the moment. Could you just check if maybe the bracket are missing when you have set your config values.

jkaberg, 2023-08-05:
@14yannick it happens also when I don't set any values in configuration.h

jkaberg, 2023-08-05:
just to be clear, this happens on development branch. master is fine, sorry if there were any confusion

14yannick, 2023-08-05:
Ok wifi handling is quite different in this branch. I haven't build this release yet. Maybe @Tysonpower can advise.

14yannick, 2023-08-07:
did you have the last version of the development because some merge took place on the 4th of august. Maybe you pull a version between two merges.

Gifford47, 2023-08-07:
@14yannick yes, we can do it. But then the default PR branch would also be the master branch which shouldn't be. Or is there a way to change that behaviour?

jkaberg, 2023-08-30:
Sorry for the late reply here, the fix in master branch fixed it 👍

commit b21a2c39 (2023-08-03): Fix Issue Loss of WIFI results in WIFI connect loop #31
Reconnect timer was cleared after reconnect
Close #31.
PR #33 (2023-08-03): Merge dev fixes and features into master
Reconnect timer was cleared after reconnect
Close #31.
closed 2023-08-30 by jkaberg