Gifford47/HCPBridgeMqtt #43: Security: Disable open SSID from the ESP32

dfluegel, 2023-09-04:
**Describe the bug**
I discovered, that both of my ESP32 are broadcasting an SSID ESP_XXXXXX. After connecting without password, I'm able to open the garage door via browser and the ip address 192.168.4.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Which motor do you have?
2. Which branch/release did you use?
> master, 4th Sept 2023
4. Which peripherals (bme280, HCsr04, ...) did you use?
> none
6. See error

**Expected behavior**
An option in the configration file to disable the SSID, otherwise if connected as client to a wifi network, the ESP shouldn't provide it's own SSID

**Screenshots**
Probably not required

**Additional context**

Gifford47, 2023-09-04:
i hope its the dev branch because the current master branch shouldn`t have an ap mode :)

@14yannick i think we should also add a AP password like ```WiFi.softAP("MY-WIFI-AP", "MYSECRETPASS");``` in the upcoming development_pref branch. The MY-WIFI-AP and MYSECRETPASS could also be changable via webif.

dfluegel, 2023-09-04:
@14yannick my bad, I have changed the branch to dev and forgot about it....

@Gifford47 Besides the fact that I was in the wrong branch, this seems to be a good idea.

@Tysonpower @14yannick  I'm closing this bug report, as this is seems to be a different issue.

closed 2023-09-04 by dfluegel

14yannick, 2023-09-04:
@Tysonpower I also already thought about it with the issue that a faulty setting resulting in the need to clear the memory from a Serial interface as it's stuck. I know that shelly for example resolve this issue detecting two successif shut down. I our case we could for example use the physical reset button. If it's pressed twice in less than 10sec then clear all preferences settings and read it again from the configuration file. 
@Gifford47 it would in any case make sens to add credential in the configuration and start the AP with a password. The only issue I see there it's that my idea is to deliver pre-build without the need to change anything in the configuration.h file. So the security issue still persist as the password of the prebuild can be found easily. 
This can be fixed adding a flag for protected mode who has to be removed. With this flag set no command can be sent to the motor.
Tell me what you think about both topics.

Tysonpower, 2023-09-04:
Adding a password will not increase security i think, button is the best to just factory reset, my boards only have a boot button but even this can be used on runtime to watch for long presses. Here is a ruff version to reset when user holds Button for over 5s and then releases it. Maybe you can use that as a starting point.

```
void reset_button_change(){
  if (digitalRead(0) == 0)
  {
    // Pressed
    resetButtonTimePressed = millis();
    Serial.print("Reset DOWN...");
  } else {
    // unpressed
    unsigned long timeNow = millis();
    int timeInSecs = (timeNow - resetButtonTimePressed) / 1000;
    if (timeInSecs > 5)
    {
      Serial.println("RESETTING....");
      //TODO: RESET to defaults
      ESP.restart();
    }
    
    Serial.print("Reset UP: ");
    Serial.println(timeInSecs);
    resetButtonTimePressed = 0;
  }
}
```

And under Setup:

```
//Add interrupts for Factoryreset over Boot button
pinMode(0, INPUT_PULLUP);
attachInterrupt(digitalPinToInterrupt(0), reset_button_change, CHANGE);
```

14yannick, 2023-09-05:
@Tysonpower I like the idea of the longpress button. I tried to implement it. See commit above could you have a look on the setup part as it cause the ESP32 to crash.

14yannick, 2023-09-05:
ok maybe I found something will check later
https://forum.arduino.cc/t/pushbutton-short-long-press-with-interrupt/667457/7

Tysonpower, 2023-09-05:
the code i posted worked on my esp32-S3, but i wasn't able to get the prefs to clear, it restarted the esp but prefs didn't clear for some reason, everything else was okay. My Code is based on the forum post you linked btw. :)