Gifford47/HCPBridgeMqtt #48: set_position does not work over MQTT

shellyeah, 2023-11-22:
**Describe the bug**
I updated to the latest version (I built the master branch, not development). I am still able to control the door. Open, Close, Stop, Light, Vent. It also correctly reads back the current position but I can't set the position manually through HomeAssistant/MQTT.
Web Interface works and I **CAN** set the position through the http endpoint. e.g: `/command?action=7&position=90`
So it must be related to the firmware, something in the MQTT part must be messed up?

**To Reproduce**
Build firmware from the current master (commit 04f1679043589d1275462a4fa0f521d6cb790c9d), install firmware and try to set position through HomeAssistant/MQTT (set_position).

1. Which motor do you have? -> Hörmann Supramatic E4
2. Which branch/release did you use? ->Master (commit 04f1679043589d1275462a4fa0f521d6cb790c9d)
3. Which peripherals (bme280, HCsr04, ...) did you use? -> None. I use tynets board however and it worked and still works fine. (also build the esp32s3_tynet binary)

In the configuration.h I only set an OTA password, disabled the access point, added my wifi and mqtt credentials, set DEVICE_ID to HCPBridge01 and HOSTNAME to HCPBRIDGE01, (In case I get ten more garages :D)

**Expected behavior**
Set the cover position through Home assistant/MQTT again as I've build a few automations around that feature.

[image]

subscribed to hormann/HCPBridge01/# and you can clearly see that the set_position command does not work while setting it through the http endpoint works as expected.
```
hormann/HCPBridge01/state {"valid":true,"doorposition":95,"lamp":"true","doorstate":"closing","detailedState":"closing","vent":"close"}
hormann/HCPBridge01/position 95
hormann/HCPBridge01/state {"valid":true,"doorposition":89,"lamp":"true","doorstate":"open","detailedState":"stopped","vent":"close"}
hormann/HCPBridge01/position 89
hormann/HCPBridge01/state {"valid":true,"doorposition":94,"lamp":"true","doorstate":"opening","detailedState":"opening","vent":"close"}
hormann/HCPBridge01/position 94
hormann/HCPBridge01/state {"valid":true,"doorposition":99,"lamp":"true","doorstate":"opening","detailedState":"opening","vent":"close"}
hormann/HCPBridge01/position 99
hormann/HCPBridge01/state {"valid":true,"doorposition":100,"lamp":"true","doorstate":"open","detailedState":"open","vent":"close"}
hormann/HCPBridge01/position 100
hormann/HCPBridge01/state {"valid":true,"doorposition":93,"lamp":"true","doorstate":"closing","detailedState":"closing","vent":"close"}
hormann/HCPBridge01/position 93
hormann/HCPBridge01/state {"valid":true,"doorposition":89,"lamp":"true","doorstate":"open","detailedState":"stopped","vent":"close"}
hormann/HCPBridge01/position 89
hormann/HCPBridge01/state {"valid":true,"doorposition":95,"lamp":"true","doorstate":"opening","detailedState":"opening","vent":"close"}
hormann/HCPBridge01/position 95
hormann/HCPBridge01/state {"valid":true,"doorposition":100,"lamp":"true","doorstate":"open","detailedState":"open","vent":"close"}
hormann/HCPBridge01/position 100
hormann/HCPBridge01/set_position 90
hormann/HCPBridge01/set_position 90
hormann/HCPBridge01/set_position 90
```

14yannick, 2023-11-22:
Was able to reproduce it. I will have a look on it.

Tysonpower, 2023-11-24:
I was also able to reproduce it, if it worked before maybe the mqtt command is not recognised correctly or something

Gifford47, 2023-11-30:
Temperatur and humidity are also not transmitted...  Only on right after boot

14yannick, 2023-11-30:
the set position work from the we Gui so it cannot be the setPosition command in the Hoermann engine funtion. The devided by 0 is because the motor has position from 0 to 200 and Home assistant has 0 to 100. To have the same scale all number from the motor are divided by 200 and the others by 100.
I'm suspect the error [here](https://github.com/Gifford47/HCPBridgeMqtt/blob/04f1679043589d1275462a4fa0f521d6cb790c9d/HCPBridgeESP32/src/main.cpp#L379C45-L379C45).
I have no issue with temperature. temperature is missing sometimes but it's when my docker reboot.
[image]

14yannick, 2024-01-05:
it's now fixed in the development [Chanel](https://github.com/Gifford47/HCPBridgeMqtt/tree/development)

closed 2024-01-05 by 14yannick

Tysonpower, 2024-01-05:
Thanks @14yannick !