Gifford47/HCPBridgeMqtt #50: Esp with ethernet

BlastyCZ, 2023-12-22:
**Is your feature request related to a problem? Please describe.**
Weak WiFi signal

**Describe the solution you'd like**
It would nice to add option to use esp32 board with ethernet for example WT32-ETH01

Gifford47, 2023-12-22:
Hi!
It would be a nice feature, but we don't have a plan for this.
The reason is that the code size would increase significantly and the ESP would have to serve the ethernet in addition to the bus communication. since we already use both cores and the core for the bus communication should not take over any other tasks, I could not guarantee that everything would run smoothly.
If the wifi signal is weak, please use a repeater.

closed 2023-12-22 by Gifford47

Eagle3386, 2024-11-02:
@Gifford47 Excuse me for kinda "reviving" this, but how about dropping WiFi for PoE?
That way, the burden you've mentioned wouldn't be imposed as it's basically an exchange of the network part.

I'd love to see this happening, even if I had to pay 50-60 €, just to not have another wireless device, but use an Ethernet connection & preferably via PoE, so that it just takes a single cable & nothing more. 😍