Gifford47/HCPBridgeMqtt #51: Can't compile code for my esp32s

alex4988, 2023-12-26:
i’m trying to compile your code for my esp32s but i receive these error:
`In file included from src/main.cpp:3:
.pio/libdeps/esp32s3_tynet/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https: [-Wcpp]
 #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https://docs.elegantota.pro/async-mode/
  ^~~~~~~
Compiling .pio\build\esp32s3_tynet\liba4b\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32s3_tynet\liba4b\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32s3_tynet\liba4b\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32s3_tynet\liba4b\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32s3_tynet\liba4b\WiFi\WiFiUdp.cpp.o
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:753:2: error: #else without #if
 #else
  ^~~~
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:755:2: error: #endif without #if
 #endif
  ^~~~~
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:1020: error: unterminated #if
 #if LWIP_IPV4 && LWIP_IPV6

Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\AsyncEventSource.cpp.o
Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\AsyncWebSocket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\WebAuthentication.cpp.o
Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\WebHandlers.cpp.o
Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\WebRequest.cpp.o
Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\WebResponses.cpp.o
Compiling .pio\build\esp32s3_tynet\lib829\ESPAsyncWebServer-esphome\WebServer.cpp.o
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:754:39: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
         return connect(IPAddress(addr.addr), port);
                                       ^~~~
                                       u_addr
*** [.pio\build\esp32s3_tynet\libfda\AsyncTCP-esphome\AsyncTCP.cpp.o] Error 1`

Why?

Thank you so much for your work

14yannick, 2023-12-26:
You qre in vs code platformio right. There clean and build again. Please add the full content of your console. Specially the beginning to see the version of the libraries used.

alex4988, 2023-12-26:
below log complete of VS:
` *  Executing task: C:\Users\Alex\.platformio\penv\Scripts\platformio.exe run --environment esp32s3_tynet 

Processing esp32s3_tynet (board: adafruit_feather_esp32s3; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Compressing WebUI...
Used environment:esp32s3_tynet
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32s3.html
PLATFORM: Espressif 32 (6.4.0) > Adafruit Feather ESP32-S3 2MB PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ArduinoJson @ 6.21.4
|-- EspSoftwareSerial @ 8.1.0
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient-esphome @ 0.8.6
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0+sha.df666c6
|-- WiFi @ 2.0.0
|-- AsyncTCP-esphome @ 2.1.0
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0
Building in release mode
Compiling .pio\build\esp32s3_tynet\libfda\AsyncTCP-esphome\AsyncTCP.cpp.o
Archiving .pio\build\esp32s3_tynet\lib6ca\libFS.a
Archiving .pio\build\esp32s3_tynet\liba4b\libWiFi.a
Archiving .pio\build\esp32s3_tynet\lib829\libESPAsyncWebServer-esphome.a
Compiling .pio\build\esp32s3_tynet\lib25f\EspSoftwareSerial\SoftwareSerial.cpp.o
Compiling .pio\build\esp32s3_tynet\libf22\Update\HttpsOTAUpdate.cpp.o
Compiling .pio\build\esp32s3_tynet\libf22\Update\Updater.cpp.o
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:753:2: error: #else without #if
 #else
  ^~~~
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:755:2: error: #endif without #if
 #endif
  ^~~~~
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:1020: error: unterminated #if
 #if LWIP_IPV4 && LWIP_IPV6

Compiling .pio\build\esp32s3_tynet\lib50d\AsyncElegantOTA\AsyncElegantOTA.cpp.o
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio/libdeps/esp32s3_tynet/AsyncTCP-esphome/src/AsyncTCP.cpp:754:39: error: 'ip_addr_t' {aka 'struct ip_addr'} has no member named 'addr'; did you mean 'u_addr'?
         return connect(IPAddress(addr.addr), port);
                                       ^~~~
                                       u_addr
In file included from .pio/libdeps/esp32s3_tynet/AsyncElegantOTA/src/AsyncElegantOTA.cpp:1:
.pio/libdeps/esp32s3_tynet/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https: [-Wcpp]
 #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https://docs.elegantota.pro/async-mode/
  ^~~~~~~
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\ConnAckPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\PingRespPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\PubAckPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\PubCompPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\PubRecPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\PubRelPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\PublishPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\SubAckPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib9b2\AsyncMqttClient-esphome\AsyncMqttClient\Packets\UnsubAckPacket.cpp.o
Compiling .pio\build\esp32s3_tynet\lib4e6\OneWire\OneWire.cpp.o
Compiling .pio\build\esp32s3_tynet\lib6b4\modbus-esp8266\Modbus.cpp.o
*** [.pio\build\esp32s3_tynet\libfda\AsyncTCP-esphome\AsyncTCP.cpp.o] Error 1
.pio/libdeps/esp32s3_tynet/OneWire/OneWire.cpp:599:22: warning: extra tokens at end of #undef directive
 #  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
                      ^
.pio/libdeps/esp32s3_tynet/OneWire/OneWire.cpp:600:20: warning: extra tokens at end of #undef directive
 #  undef interrupts() portEXIT_CRITICAL(&mux);}
                    ^
================================================================================================== [FAILED] Took 8.49 seconds ==================================================================================================

[... cut ...]
-------------  --------  ------------
esp32s3_tynet  FAILED    00:00:08.488
============================================================================================ 1 failed, 0 succeeded in 00:00:08.488 ============================================================================================ 

 *  The terminal process "C:\Users\Alex\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32s3_tynet'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

14yannick, 2023-12-26:
it's coming from the  AsyncTCP-esphome @ 2.1.0 do you have it as dependency in your platform.io file.
it should not be there as it's automatically added in version AsyncTCP-esphome @ 2.0.0 by an dependency.

14yannick, 2023-12-26:
A fix of the asynctcp is in the pipeline https://github.com/esphome/AsyncTCP/pull/6

alex4988, 2023-12-27:
Thank you so much, i force to use 2.0 library in platformio.ini below code:
`lib_deps = 
	esphome/ESPAsyncWebServer-esphome
	**esphome/AsyncTCP-esphome@2.0.0**
	bblanchon/ArduinoJson
	plerup/EspSoftwareSerial
	ayushsharma82/AsyncElegantOTA
	ottowinter/AsyncMqttClient-esphome
	paulstoffregen/OneWire
	https://github.com/emelianov/modbus-esp8266`

closed 2023-12-27 by 14yannick