Gifford47/HCPBridgeMqtt #60: MQTT Connection

mic9119, 2024-01-31:
Hello,

I downloaded before 4 days all files and programmed my ESP. 

I Have installed Mosquitto Broker on my PC. I added the user from the ESP. The Server IP is the IP from my PC. But on TeraTerm my ESP say "Connecting to MQTT..."

The Mosquitto Broker is running. I tested this with MQTT.fx an MQTT Explorer.

Best regards

mic9119, 2024-01-31:
src/main.cpp: In function 'void onMqttConnect(bool)':
src/main.cpp:683:35: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       sendDebug("ResetReason", val);
                                   ^
src/main.cpp: In function 'void mqttTaskFunc(void*)':
src/main.cpp:699:19: error: too few arguments to function 'void sendDebug(char*, String)'
         sendDebug();
                   ^
src/main.cpp:402:6: note: declared here
 void sendDebug(char *key, String value)

There are errors

14yannick, 2024-01-31:
Replace [this](https://github.com/Gifford47/HCPBridgeMqtt/blob/edda64c082be56d0efae877fe3333b2a830f681d/HCPBridgeESP32/src/main.cpp#L699) line with sendDebug("engineMes",hoermannEngine->state->debugMessage);
Also move it one line up before the line 
hoermannEngine->state->clearDebug();

mic9119, 2024-01-31:
I get this Info via serial Monitor:

Connecting to MQTT...
[  3579][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  4079][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  4579][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  5079][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  5579][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  6080][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  6580][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  7080][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  7580][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  8080][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  8580][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  9081][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[  9581][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 10081][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 10581][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 11081][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 11581][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 12082][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 12582][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 13082][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 13582][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 14082][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 14582][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 15083][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 15583][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 16083][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 16583][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 17083][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 17583][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 18084][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 18584][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 19084][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 19584][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 20084][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 20584][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
[ 21085][W][AsyncTCP.cpp:986] _poll(): pcb is NULL
Disconnected from MQTT. reason : TCP_DISCONNECTED

14yannick, 2024-01-31:
Ok please send the terminal output of the build command to see the version of libraries used for build

mic9119, 2024-01-31:
Library Manager: Installing dok-net/ghostl @ ^1.0.0
Library Manager: Warning! Could not install `{'owner': 'dok-net', 'name': 'ghostl', 'version': '^1.0.0'}` dependency for the`EspSoftwareSerial` package
Verbose mode can be enabled via `-v, --verbose` option
Compressing WebUI...  
Used environment:esp32
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (6.5.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- AsyncTCP-esphome @ 2.1.0+sha.ceefe2c
|-- ArduinoJson @ 7.0.2
|-- EspSoftwareSerial @ 8.2.0
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient-esphome @ 0.8.6
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0
|-- ghostl @ 1.0.1+sha.52a206b
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0

do you mean this?

mic9119, 2024-01-31:
The esp ist connectet to the wifi

14yannick, 2024-01-31:
try to change the libs in platformio.ini with this
```
lib_deps = 
	esphome/ESPAsyncWebServer-esphome
	;https://github.com/HeMan/AsyncTCP#CI-and-fix ;async tcp issue fix 2.1.0
	bblanchon/ArduinoJson
	plerup/EspSoftwareSerial
	ayushsharma82/AsyncElegantOTA
	heman/AsyncMqttClient-esphome
	paulstoffregen/OneWire
	emelianov/modbus-esp8266
	maxgerhardt/ghostl ; << Fix missing circular_queue.h
```

mic9119, 2024-02-01:
I get the same log...

14yannick, 2024-02-01:
start an neu platformio console 
[image]
and type `pio pkg uninstall` and after `pio pkg install`
Your build should look like something like this ````
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ArduinoJson @ 7.0.2
|-- EspSoftwareSerial @ 8.2.0
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient-esphome @ 2.0.0
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0
|-- ghostl @ 1.0.1
|-- home-assistant-integration @ 2.0.0
|-- WiFi @ 2.0.0
|-- AsyncTCP-esphome @ 2.1.1
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0`

```

mic9119, 2024-02-01:
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ArduinoJson @ 7.0.2
|-- EspSoftwareSerial @ 8.2.0
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient-esphome @ 0.8.6-------------
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0
|-- ghostl @ 1.0.1+sha.52a206b-------
-----------
|-- WiFi @ 2.0.0
|-- AsyncTCP-esphome @ 2.1.0+sha.ceefe2c-------
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0

AsyncMqtt is very old
ghostl is there a difference?
home-assistang is lost
AsyncTCP is there a difference?

mic9119, 2024-02-01:
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ArduinoJson @ 7.0.2
|-- EspSoftwareSerial @ 8.2.0
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient-esphome @ 2.0.0
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0
|-- ghostl @ 1.0.1
|-- WiFi @ 2.0.0
|-- AsyncTCP-esphome @ 2.1.1
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0

Now it looks like the dependencys from you
But the los is always the same