Gifford47/HCPBridgeMqtt #61: Temperature

mic9119, 2024-01-31:
Is there chance to get the temperature from the dht sensor in the /status page?

Gifford47, 2024-01-31:
Yes, I'm already working on it...

Gifford47, 2024-02-03:
Look at the latest dev version. Temperature and humidity are shown in the UI if sensor is available. It's the first iteration for the UI. If i have time enough I'll rework the display setting.

closed 2024-02-03 by Gifford47

mic9119, 2024-02-05:
Hey,
i flashed my ESP32 in my office. The esp in not connected to the drive. I get the following picture:
[image]

Gifford47, 2024-02-05:
look at system infomation->door status. do you get a json with "temp" attribute? if yes, please press F12 (on desktop chrome browser), reload the page and look if there's an extra section before the door buttons.