Gifford47/HCPBridgeMqtt #63: No possibility to change position via mqtt when using IOBroker

malu254, 2024-02-02:
The Communication between HCP and IOBroker is working.

Somehow IOBroker isnt able to toggle movement of the garage door.

Iobroker needs a true / false mqtt state to read or write data via mqtt

Control via the Web interface works well

esusxunil, 2024-06-04:
Same here, I get all status updates through MQTT into ioBroker.

As a workaround I just trigger the commands through http.

Light: http://[ip]/command?action=5
Stop: http://[ip]/command?action=2
Up: http://[ip]/command?action=1
Down: http://[ip]/command?action=0

14yannick, 2024-08-22:
Maybe the esphome [port ](https://github.com/mapero/esphome-hcpbridge) of this project can help. As I saw there is an Esphome integration in IObroker.

closed 2025-01-14 by Gifford47