Gifford47/HCPBridgeMqtt #72: Support Client ID for MQTT connection

Autonoma-Bauernfeind, 2024-06-22:
**Is your feature request related to a problem? Please describe.**
In the MQTT definition a client ID shall be provided. Also a lot of MQTT broker need a unique Client ID to be able to connect-

**Describe the solution you'd like**
An additional field or using the already existing field Device ID as client ID?

**Describe alternatives you've considered**
There is no alternative to that

Gifford47, 2024-06-25:
hi!
it´s in the current dev branch, see #71 . i`ll merge it into master soon.

closed 2024-06-25 by Gifford47

Tysonpower, 2024-07-02:
@Autonoma-Bauernfeind the ID_LENGTH is not defined in the config, what should be the default value for that?

Gifford47, 2024-07-02:
ok, my configuration wasn't merged, sry. the length could be 16 chars. Do you push a tested PR?

Tysonpower, 2024-07-02:
no didn't do a pr, just hardcoded it locally so i can build, so feel free to push your config