Gifford47/HCPBridgeMqtt #78: Tynet_BME280 Unable to build

Osjuh, 2024-07-06:
When I try to build the Board in VStudio with the ESP32S3_Tynet_BME280 board, I get the error in main.php and thefore I'm not able to upload. the ESP32S3_Tyenet works without issues but then I've no temp and pressure information, See printscreen.

**To Reproduce**
Steps to reproduce the behavior:
1. Which motor do you have?
Supramatic 4
2. Which branch/release did you use?
2024.07.03
3. Which peripherals (bme280, HCsr04, ...) did you use?
BME280
4. See error

` *  Executing task: platformio run --environment esp32s3_tynet_BME280 

Processing esp32s3_tynet_BME280 (board: adafruit_feather_esp32s3; platform: espressif32@6.5.0; framework: arduino)
----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Compressing WebUI...
Used environment:esp32s3_tynet_BME280
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/adafruit_feather_esp32s3.html
PLATFORM: Espressif 32 (6.5.0) > Adafruit Feather ESP32-S3 2MB PSRAM
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- AsyncTCP-esphome @ 2.1.0+sha.ceefe2c
|-- ArduinoJson @ 7.1.0
|-- EspSoftwareSerial @ 8.2.0
|-- AsyncElegantOTA @ 2.2.8
|-- AsyncMqttClient-esphome @ 0.8.6
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0
|-- ghostl @ 1.0.1+sha.52a206b
|-- Adafruit BME280 Library @ 2.2.4
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0
|-- Wire @ 2.0.0
|-- Adafruit Unified Sensor @ 1.1.14
|-- Adafruit BusIO @ 1.16.1
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio/build/esp32s3_tynet_BME280/src/main.cpp.o
In file included from src/main.cpp:3:
.pio/libdeps/esp32s3_tynet_BME280/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https: [-Wcpp]
 #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https://docs.elegantota.pro/async-mode/
  ^~~~~~~
Archiving .pio/build/esp32s3_tynet_BME280/lib369/libWiFi.a
Indexing .pio/build/esp32s3_tynet_BME280/lib369/libWiFi.a
Compiling .pio/build/esp32s3_tynet_BME280/lib15d/ESPAsyncWebServer-esphome/WebServer.cpp.o
Compiling .pio/build/esp32s3_tynet_BME280/lib8b2/ghostl@src-82472abe6282970571c43e774c39a00f/FastScheduler.cpp.o
Compiling .pio/build/esp32s3_tynet_BME280/libff2/EspSoftwareSerial/SoftwareSerial.cpp.o
In file included from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:88:25: warning: 'DynamicJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
     DynamicJsonDocument _jsonBuffer;
                         ^~~~~~~~~~~
In file included from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.hpp:56,
                 from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.h:9,
                 from .pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:37,
                 from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson/compatibility.hpp:125:58: note: declared here
 class ARDUINOJSON_DEPRECATED("use JsonDocument instead") DynamicJsonDocument
                                                          ^~~~~~~~~~~~~~~~~~~
In file included from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h: In constructor 'AsyncJsonResponse::AsyncJsonResponse(bool, size_t)':
.pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:110:47: warning: 'ArduinoJson::V710PB22::JsonArray ArduinoJson::V710PB22::JsonDocument::createNestedArray()' is deprecated: use add<JsonArray>() instead [-Wdeprecated-declarations]
         _root = _jsonBuffer.createNestedArray();
                                               ^
In file included from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.hpp:33,
                 from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.h:9,
                 from .pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:37,
                 from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:326:13: note: declared here
   JsonArray createNestedArray() {
             ^~~~~~~~~~~~~~~~~
In file included from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:112:48: warning: 'ArduinoJson::V710PB22::JsonObject ArduinoJson::V710PB22::JsonDocument::createNestedObject()' is deprecated: use add<JsonObject>() instead [-Wdeprecated-declarations]
         _root = _jsonBuffer.createNestedObject();
                                                ^
In file included from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.hpp:33,
                 from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.h:9,
                 from .pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:37,
                 from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:346:14: note: declared here
   JsonObject createNestedObject() {
              ^~~~~~~~~~~~~~~~~~
In file included from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h: In member function 'virtual void AsyncCallbackJsonWebHandler::handleRequest(AsyncWebServerRequest*)':
.pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:224:39: warning: 'DynamicJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
         DynamicJsonDocument jsonBuffer(this->maxJsonBufferSize);
                                       ^
In file included from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.hpp:56,
                 from .pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson.h:9,
                 from .pio/libdeps/esp32s3_tynet_BME280/ESPAsyncWebServer-esphome/src/AsyncJson.h:37,
                 from src/main.cpp:5:
.pio/libdeps/esp32s3_tynet_BME280/ArduinoJson/src/ArduinoJson/compatibility.hpp:125:58: note: declared here
 class ARDUINOJSON_DEPRECATED("use JsonDocument instead") DynamicJsonDocument
                                                          ^~~~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32s3_tynet_BME280/libe0d/Update/HttpsOTAUpdate.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:1079:39: error: 'doc' was not declared in this scope
                 JsonObject sensors  = doc.createNestedObject("sensors");
                                       ^~~
Compiling .pio/build/esp32s3_tynet_BME280/libe0d/Update/Updater.cpp.o
Compiling .pio/build/esp32s3_tynet_BME280/libc59/AsyncElegantOTA/AsyncElegantOTA.cpp.o
In file included from .pio/libdeps/esp32s3_tynet_BME280/AsyncElegantOTA/src/AsyncElegantOTA.cpp:1:
.pio/libdeps/esp32s3_tynet_BME280/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https: [-Wcpp]
 #warning AsyncElegantOTA library is deprecated, Please consider moving to newer ElegantOTA library which now comes with an Async Mode. Learn More: https://docs.elegantota.pro/async-mode/
  ^~~~~~~
Compiling .pio/build/esp32s3_tynet_BME280/libc84/AsyncMqttClient-esphome/AsyncMqttClient.cpp.o
*** [.pio/build/esp32s3_tynet_BME280/src/main.cpp.o] Error 1
================================= [FAILED] Took 3.09 seconds =================================

Environment           Status    Duration
--------------------  --------  ------------
esp32s3_tynet_BME280  FAILED    00:00:03.087
=========================== 1 failed, 0 succeeded in 00:00:03.087 ===========================

 *  The terminal process "platformio 'run', '--environment', 'esp32s3_tynet_BME280'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
`

**Expected behavior**
Build successfull, so I can upload it to the board

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Gifford47, 2024-07-06:
There's a bug in current master I will solve in the current days. Please use current dev branch instead.

Osjuh, 2024-07-06:
> There's a bug in current master I will solve in the current days. Please use current dev branch instead.

This worked for me. Thank you.
Never build the board so fast as with the dev branch.
Thank you and keep up the good work.

14yannick, 2024-07-06:
I think it should be root instead of doc as else the values of the sensor will not be in the response.
[image]

Gifford47, 2024-07-08:
closed with release 2024.07.08

closed 2024-07-08 by Gifford47