Gifford47/HCPBridgeMqtt #79: AsyncElegantOTA Update breaks build

ProtoxiDe22, 2024-07-07:
**Describe the bug**
a just released version of AsyncElegantOTA breaks the build, the 2.2.9 version, released between yesterday and today, intentionally throws an error and fails the build because the library is deprecated: this is line 4 of AsyncElegantOTA.h
```c
#error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https://docs.elegantota.pro/async-mode/ 
```

**To Reproduce**
Steps to reproduce the behavior:
try to build the project on a fresh repo (you need to force platformio to redownload/update dependencies if you already have the build env set up)

**Expected behavior**
project should build

**Additional context**
build log:
``` 
*  Executing task: C:\Users\REDACTED\.platformio\penv\Scripts\platformio.exe run --environment esp32 

Processing esp32 (platform: espressif32@6.5.0; board: nodemcu-32s; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Compressing WebUI...
Used environment:esp32
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (6.5.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- AsyncTCP-esphome @ 2.1.0+sha.ceefe2c
|-- ArduinoJson @ 7.1.0
|-- EspSoftwareSerial @ 8.2.0
|-- AsyncElegantOTA @ 2.2.9
|-- AsyncMqttClient-esphome @ 0.8.6
|-- OneWire @ 2.3.8
|-- modbus-esp8266 @ 4.1.0
|-- ghostl @ 1.0.1+sha.52a206b
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0
Building in release mode
Compiling .pio\build\esp32\src\main.cpp.o
Building .pio\build\esp32\bootloader.bin
Generating partitions .pio\build\esp32\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32\libc49\AsyncTCP-esphome@src-ab8ca2c2ec4b0057e959fb2d80027908\AsyncTCP.cpp.o
Compiling .pio\build\esp32\lib5f9\FS\FS.cpp.o
Compiling .pio\build\esp32\lib5f9\FS\vfs_api.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32\lib80b\WiFi\WiFiServer.cpp.o
In file included from src/main.cpp:3:
.pio/libdeps/esp32/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: error: #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https:
[... 51 lines cut ...]
In file included from .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson.hpp:56,
                 from .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson.h:9,
                 from .pio/libdeps/esp32/ESPAsyncWebServer-esphome/src/AsyncJson.h:37,
                 from src/main.cpp:5:
.pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/compatibility.hpp:125:58: note: declared here
 class ARDUINOJSON_DEPRECATED("use JsonDocument instead") DynamicJsonDocument
                                                          ^~~~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32\lib6e1\Update\HttpsOTAUpdate.cpp.o
Compiling .pio\build\esp32\lib6e1\Update\Updater.cpp.o
Compiling .pio\build\esp32\libb5b\AsyncElegantOTA\AsyncElegantOTA.cpp.o
*** [.pio\build\esp32\src\main.cpp.o] Error 1
In file included from .pio/libdeps/esp32/AsyncElegantOTA/src/AsyncElegantOTA.cpp:1:
.pio/libdeps/esp32/AsyncElegantOTA/src/AsyncElegantOTA.h:4:2: error: #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https:
 #error AsyncElegantOTA library is deprecated, Please consider migrating to newer ElegantOTA library which now comes with an async mode. Learn More: https://docs.elegantota.pro/async-mode/
  ^~~~~
*** [.pio\build\esp32\libb5b\AsyncElegantOTA\AsyncElegantOTA.cpp.o] Error 1
========================================================================================================= [FAILED] Took 7.68 seconds =========================================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:00:07.675
=================================================================================================== 1 failed, 0 succeeded in 00:00:07.675 =================================================================================================== 

 *  The terminal process "C:\Users\REDACTED\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32'" terminated with exit code: 1. ****
```

14yannick, 2024-07-08:
AsyncEleganteOTA is not available anymore maybe an easy change in the platformio.ini file will solve the issue migrating to EleganteOTA in async mode.
https://docs.elegantota.pro/async-mode/
Or maybe it will fail as the version of the dependency are not met...

Gifford47, 2024-07-08:
closed with release 2024.07.08

closed 2024-07-08 by Gifford47