Gifford47/HCPBridgeMqtt #88: configuration is lost after a few minutes (2024.07.08)

webflash-aj, 2024-08-08:
**Describe the bug**
when I flash my ESP with the new firmware, nothing works after a few minutes. after rebooting the ESP, my entire configuration is lost (WiFi & MQTT).

the old version 0.0.6.1 does not show this behavior.

any ideas?

**To Reproduce**
- Flash ESP
- Configure WiFi and MQTT
- Wait a few minutes
- ESP is no longer responsive
- The settings are lost after a reboot (power cycle)

ESP is not connected to the Supramatic

Gifford47, 2024-08-08:
Does it also happen when connected to the motor?

webflash-aj, 2024-08-08:
unfortunately yes

webflash-aj, 2024-08-19:
@burmistrzak 
My ESP says goodbye after a few minutes with the new firmware. What I have not tried is to compile the repo myself.

If anyone would like me to, I would be happy to test it. Unfortunately I don't have the experience for debugging.

dev-strom, 2024-08-19:
I can commit this behavior. I'm using an ESP32 with the BME280 sensor.
The first time I run this software it lost the configuration multiple times a day. I tried some stuff and installed an additional wifi repeater to make the wifi connection more reliable, After that the hcp bridge ran a few weeks (or was it month?) without any problems. But last week it lost its configuration again one time.
I'm not sure, what the reason is. But it looks to me, it could be an issue with a lost wifi connection.

BradleyFord, 2024-08-19:
It seems much easier to just switch the ESPHome version of the code, then all of these issues disappear as that is all standard functionality

Tysonpower, 2024-08-19:
Currently on vacation so i can't test that, but i wonder if this only happens on the BME and DHT22 Sensor enabled versions, sounds like that to me currently. Still annoying and this should not happen, i think i will make a stable fork for my boards if the repo is that unstable. Maybe i switch to esphome as well, didnt try the esphome version yet. @BradleyFord Do you have a link t the esphome version?

BradleyFord, 2024-08-19:
The ESPHome fork is here, essentially it is build off the good work ok people in this thread. 
But the core elements have been extracted and is used within ESPHome, hence then you get the ESPHome stability and extras 

https://github.com/mapero/esphome-hcpbridge

webflash-aj, 2024-08-19:
@burmistrzak 
I tested with an esp32 and just the rs468-module connected, nothing else. downloaded the esp32.bin binary from github.
Also tested esp32_BME280_HCSR04.bin (without connecting a sensor) with the same results.

Tysonpower, 2024-08-19:
I will investigate into the issue in two weeks when i'm back home, if you find the issue let me know.

webflash-aj, 2024-08-19:
@burmistrzak 
No, "normal" USB-C ESP32-Board (see attached Photo for my setup) powered over USB-C. Old Firmware have no porblems. WiFI should be also fine (Strong Signal). AP is just arround the corner.

[image]

Tysonpower, 2024-08-19:
> @webflash-aj Hmm, I now believe the ESP isn't actually resetting, because `ESP.restart();` would certainly brick the motor and that didn't happen (at least in my case).
> 
> https://github.com/Gifford47/HCPBridgeMqtt/blob/514164d3409782c7c6f95bce8b55b10a8383fc31/HCPBridgeESP32/src/preferencesKeys.h#L205-L209
> 
> IMHO, it's more likely that `firstStart` somehow gets flipped back on...
> 
> https://github.com/Gifford47/HCPBridgeMqtt/blob/514164d3409782c7c6f95bce8b55b10a8383fc31/HCPBridgeESP32/src/preferencesKeys.h#L137
> 
> Not only network preferences are gone, but also any sensor adjustments made previously.

Yeah this sounds to be the issue i guess, but why is the big question, i can't find any reference in the code so far how this can get reset by accident. But i'm happy it's not a PCb issue with my board :)

Tysonpower, 2024-08-19:
This is cool @burmistrzak ! Thanks for the work! 

@14yannick looks like you are actively working on the Fork, is it possible to make a all in one sensor firmware with esphome as well? Maybe we should focus on your Fork?

14yannick, 2024-08-20:
Hi, I also moved a couple month to the esphome port. The framework really help in the maintainability and reduce by a big margin the needed code.
@burmistrzak can have a look in my [example ](https://github.com/14yannick/esphome-hcpbridge/blob/main/example_hcpbridge.yaml)yaml there I have implement the vent function with a template. if you want to do the same for the half open it would look like something like this.

```
switch:
  - platform: template
    id: switch_half
    name: "Open to Half"
    lambda: |-
      return (id(garagedoor_cover).get_cover_state() == HoermannState::HALFOPEN);
    optimistic: True
    turn_on_action:
      - lambda: |-
          if (id(garagedoor_cover).get_cover_state() != HoermannState::HALFOPEN) {
            id(garagedoor_cover).on_go_to_half();
          }
    turn_off_action:
      #without wait device not booting
      - wait_until:
          condition:
            binary_sensor.is_on: hcpbridge_id
      - if:
          condition:
            lambda :  |-
              return (id(garagedoor_cover).position != COVER_CLOSED);
          then:
            - cover.close: garagedoor_cover
```
@Tysonpower if you want to get the door position in percent you can do it by adding a sensor like this.
It's untested but you can play around with.
```
sensor:
  - platform: template
    name: cover percent
    id: cover_percent
    icon: 'mdi:percent-box-outline'
    unit_of_measurement: '%'
    update_interval: 5s
    filters:
      - lambda: |-
          static std::string last;
          if (x == last)
            return {};
          last = x;
          return x;
    lambda: |-
      return {id(garagedoor_cover).position};
```
Actually I'm working to get the logic of the [relay ](https://github.com/14yannick/esphome-hcpbridge/tree/relay)state you can configure to get the prewarn signal or other state based on the settings of menu 30 of the motor. This will solve this [issue](https://github.com/Gifford47/HCPBridgeMqtt/discussions/44).

webflash-aj, 2024-08-21:
@14yannick  sorry to reuse this thread, but could you help me to get esphome running on HA?

I will get this error, when compiling with your example-config on github:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/eg-out-garage.yaml...
INFO Updating https://github.com/mapero/esphome-hcpbridge.git@None
INFO Generating C++ source...
INFO Compiling app...
Processing hcpbridge (board: adafruit_feather_esp32s3; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- modbus-esp8266 @ 4.1.0
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.4
Compiling .pioenvs/hcpbridge/src/main.cpp.o
Archiving .pioenvs/hcpbridge/libd16/libAsyncTCP-esphome.a
Compiling .pioenvs/hcpbridge/lib932/modbus-esp8266/Modbus.cpp.o
Compiling .pioenvs/hcpbridge/lib932/modbus-esp8266/ModbusRTU.cpp.o
/config/esphome/eg-out-garage.yaml: In lambda function:
/config/esphome/eg-out-garage.yaml:61:33: error: 'class esphome::hcpbridge::HCPBridgeCover' has no member named 'get_relay_state'; did you mean 'restore_state_'?
       return (id(garagedoor_cover).get_relay_state());
                                 ^~~~~~~~~~~~~~~
                                 restore_state_
/config/esphome/eg-out-garage.yaml: In lambda function:
/config/esphome/eg-out-garage.yaml:71:25: error: 'class esphome::hcpbridge::HCPBridgeCover' has no member named 'set_light_state'; did you mean 'publish_state'?
         id(garagedoor_cover).set_light_state(state);
                         ^~~~~~~~~~~~~~~
                         publish_state
/config/esphome/eg-out-garage.yaml: In lambda function:
/config/esphome/eg-out-garage.yaml:101:33: error: 'class esphome::hcpbridge::HCPBridgeCover' has no member named 'get_cover_state'; did you mean 'get_component_state'?
       switch (id(garagedoor_cover).get_cover_state()) {
                                 ^~~~~~~~~~~~~~~
                                 get_component_state
/config/esphome/eg-out-garage.yaml: In lambda function:
/config/esphome/eg-out-garage.yaml:141:33: error: 'class esphome::hcpbridge::HCPBridgeCover' has no member named 'get_cover_state'; did you mean 'get_component_state'?
       return (id(garagedoor_cover).get_cover_state() == HoermannState::VENT);
                                 ^~~~~~~~~~~~~~~
                                 get_component_state
/config/esphome/eg-out-garage.yaml: In lambda function:
/config/esphome/eg-out-garage.yaml:145:29: error: 'class esphome::hcpbridge::HCPBridgeCover' has no member named 'get_cover_state'; did you mean 'get_component_state'?
           if (id(garagedoor_cover).get_cover_state() != HoermannState::VENT) {
                             ^~~~~~~~~~~~~~~
                             get_component_state
/config/esphome/eg-out-garage.yaml: In lambda function:
/config/esphome/eg-out-garage.yaml:43:33: error: 'class esphome::hcpbridge::HCPBridgeCover' has no member named 'get_light_state'; did you mean 'publish_state'?
       return (id(garagedoor_cover).get_light_state());
                                 ^~~~~~~~~~~~~~~
                                 publish_state
*** [.pioenvs/hcpbridge/src/main.cpp.o] Error 1
========================= [FAILED] Took 49.02 seconds =========================

```

14yannick, 2024-08-21:
@webflash-aj  in the yaml replace 

```
external_components:
    source: github://mapero/esphome-hcpbridge
    refresh: 0s # Ensure you always get the latest version

```
With

```
external_components:
    source: github://14yannick/esphome-hcpbridge
    refresh: 0s # Ensure you always get the latest version

```

webflash-aj, 2024-08-21:
> source: github://14yannick/esphome-hcpbridge

Thanks for your fast Feedback.. just found the "Problem" one Second before your Post.. :-)
Didn't realize the wrong repo in the yaml.

webflash-aj, 2024-08-21:
@14yannick : maybe you will also change the board to "esp32dev". It will make it easier for beginners/dummies like me..  in the example.xml

webflash-aj, 2024-08-22:
@14yannick : 
maybe you want also to change the RX-PIN to 16 (or just left somewhere a hint) i use de default wireing 
 rx_pin: 16 # optional, default=18
 tx_pin: 17 # optional, default=17

@burmistrzak & @14yannick : 
thanks for the esphome-fork.. i will give it a try.. it's working and it looks good so far.

14yannick, 2024-08-22:
A lot of people use an esp s3 board specially the one from tynet. For them the board from and pin in the yaml are correct. I think I will comments the stuff and add some more explanation.

Gifford47, 2024-11-22:
please, could someone try this binary (based on v0.0.6.1)? i can't reproduce this issue on my setup.
[firmware.zip](https://github.com/user-attachments/files/17868271/firmware.zip)

webflash-aj, 2024-11-22:
I switched to the esphome version and since then it has been running smoothly again.

I'll try to find time for it at the weekend.
Should I also test version 0.0.7.1?

Gifford47, 2024-11-22:
ok, thx!
0.0.7.1 should be buggy also, but you can try. I think it has something to do with the used/changed libs after changing to ElegantOTA...

webflash-aj, 2024-11-29:
@Gifford47 : Sorry for late reply. i tried to test the new firmware. so first i flashed the old firmware and waited for a reset. but i could not reproduce the behavior i had at my inital post. in the meantime my hcp-bridge is working with esphome whitout issues, and i don't wan't to unmount things and reflash my working enviroment. hopefully you can understand..

Gifford47, 2025-01-14:
should be solved with latest version. I'll close this issue as solved. If Issue still persists, please reopen

closed 2025-01-14 by Gifford47