Gifford47/HCPBridgeMqtt #97: Invalid packets are sent to MQTT

alexnederlof, 2024-11-24:
**Describe the bug**
- Device marks its MQTT as offline
- Availability on MQTT is therefore also set to false

My Mosquitto server says (in a loop):

```
452705: New client connected from 10.42.0.0:43537 as ESP-18114243CA48 (p2, c1, k15, u'gate').                                                                                                                
1732452710: Client ESP-18114243CA48 disconnected due to malformed packet.                                                                                                                                          1732452723: New client connected from 10.42.0.0:54130 as ESP-18114243CA48 (p2, c1, k15, u'gate').                                                                                                                 1732452725: Client ESP-18114243CA48 disconnected due to malformed packet.                                                                                                                                          1732452730: New client connected from 10.42.0.0:47305 as ESP-18114243CA48 (p2, c1, k15, u'gate').                                                                                                                1732452733: Client ESP-18114243CA48 disconnected due to malformed packet.                                                                                                                                          ```

**To Reproduce**
Steps to reproduce the behavior:
1. Which motor do you have? Supramatic E4
2. Which branch/release did you use? Version 0.0.7.1
3. Which peripherals (bme280, HCsr04, ...)  [RS485 Bridge - DIY Hörmann MQTT Adapter - ESP32-S3 Dev Board](https://shop.tynet.eu/en/)
4. See error

**Expected behavior**
Stable connection to MQTT

closed 2024-11-24 by alexnederlof

Gifford47, 2024-11-24:
can't reproduce it. It seems that's another client or saftware, isn't it?

alexnederlof, 2024-11-24:
Yes, I opened on the wrong repo, sorry. Needed to be https://github.com/Tysonpower/HCPBridgeMqtt_tynet

machine-bit, 2025-06-28:
I am getting this same issue @Gifford47 on the board you have sent me. It’s not detecting the device in Home Assistant either 

[image]

Gifford47, 2025-06-28:
Please enable verbosity level to debug and resend log on connect

Tysonpower, 2025-06-28:
> I am getting this same issue [@Gifford47](https://github.com/Gifford47) on the board you have sent me. It’s not detecting the device in Home Assistant either
> 
> [image]

Check if your DeviceID is to long, MQTT doesent like IDs that are too long. But verbosity level in the log could help as well to understand the issue better as gifford mentioned.

Tysonpower, 2025-06-28:
> [@Tysonpower](https://github.com/Tysonpower) I did try and shorten the deviceid, but maybe not enough. It seems a bit happier in the logs, just need to wait for it to show up in auto discovery now. Thanks

If HA doesent find it but in MQTT it shows up correctly maybe restart HA to make auto discovery happy again :)