Gifford47/HCPBridgeMqtt #105: Usage of Inputs with PCP

P1Rebo, 2025-02-22:
I bought the pre-built board by @Gifford47. It works very smooth by using MQTT!

Now I'd like to connect everything to my electric installation and the wall-mounted buttons. In my previous setting the Hörmann Supramatic 3 Garage Door was controlled via Relais signal to the Hörmann UAP.

Is there any way to reuse the signal for the pre-build board?

Many thanks in advance!

14yannick, 2025-03-07:
I dont know if the pre-build board has some GPIO pin available. If yes it can be connected to your relay and used in [esphome ](https://github.com/14yannick/esphome-hcpbridge) port  as GPIO Switch to trigger action.
If you provide me more information about your available GPIO on your board and how your hardware setup looks like (relay, button,...)  I can help you to create a build for your needs.

Gifford47, 2025-03-07:
Yes,my board has two independent inputs and two independent outputs available. They are already connected. You can drive "any" load you want, f.e. two Relais.
The logic depends on the user and has currently build by user.

Try to describe what you want to achieve as clear as possible.

14yannick, 2025-03-08:
ok so you have already a relay to isolate the 24v from the board. Then you you can connect a ground an available GPIO pin to the output of the relay.
In esphome then you can have a [sensor](https://esphome.io/components/binary_sensor/gpio.html) to know if the button is pressed or not a make an automation based on it.
Here an example.
```YAML
binary_sensor:
  pin:
    number: your pin number
    inverted: true
    mode:
      input: true
      pullup: true
  name: "button pressed"
    on_press:
      then:
        - cover.toggle: garagedoor_cover
```
I think it's clear but don't connect 24v to a pin of the ESP as it will fry immediately.

P1Rebo, 2025-03-08:
Thanks @14yannick. Yes, it potential free connection, so I can use the pins to connect GPIO + GND.