Gifford47/HCPBridgeMqtt #106: Lost connection after certain time

P1Rebo, 2025-03-23:
I have the Gifford47 board up and running. However, after a certain time, the connection gets interrupted... The local Wifi seems stable. It's directly connected to the Supramatic 4. Some red led stays on. Thanks for any help.

Gifford47, 2025-03-24:
reconnects it automatically? which led stays on? please check your wifi quality where your board is placed

P1Rebo, 2025-03-24:
Wifi checked. Works fine.
Tx is blinking fast, rx is on / sometimes blinking. Other then that no led is on. No reconnect automatically. Only after switch off / on

Gifford47, 2025-03-26:
ok, all works as expected except for the reconnect. It should reconnect every 2s. Do you have the opportunity to connect to the serial line while the wifi is interrupted?

P1Rebo, 2025-03-28:
I could but which pins are those and how?

Gifford47, 2025-03-28:
you need a rs232-ttl converter and the pins marked in red:

[image]

P1Rebo, 2025-03-28:
OK, is there any more documentation on that? I don't understand why there is a RS232 plug required at all. Is there anything I can connect to my Mac with Mini USB or USB-C? The connection runs for a while before it breaks. Maybe there is a log I can read out in case it's stable?

Gifford47, 2025-03-29:
Yes, technically it's possible but I would not recommend it because on some circumstances two WiFi networks could have small differences between WiFi signal strength. Maybe for a short moment. That wouldn't fix the problem.

Therefore please create different ssids for different APs.

closed 2025-04-01 by Gifford47

machine-bit, 2025-07-07:
I am having a similar issue @P1Rebo  are you using unifi kit by any chance?

Gifford47, 2025-07-08:
Partly yes. It does check for the WiFi you entered in the config and reconnects if it's available again.
It does not check for other WiFi stas.

machine-bit, 2025-07-08:
@Gifford47  if I order one of these will I be able to see what’s it’s doing? Only way to get it working again at the moment is to power cycle the garage motor. 

https://amzn.eu/d/6lL9GSb

P1Rebo, 2025-07-08:
@machine-bit I'm using Fritz Box 7590 with a Wifi Repeater. I'm using @Gifford47's board and created a separate SSID. Still, it disconnects multiple times a day and I need to power cycle it (having a ZigBee power plug in between). 

It's still not perfect, but it works with workaround and I don't have time to debug everything (that's why I bought the board) :-)

P1Rebo, 2025-07-08:
@Gifford47 I'm running version 0.0.7.3

P1Rebo, 2025-08-02:
Hi @Gifford47  - Why was this issue closed? Is there any fix? I'm still always rebooting in case of disconnect.